FlightOne/FalcoX-Issue-Tracker #1: App need to be restart to find FC

ElMute, 2020-03-01:
I need relaunch the App after each action, 
for exemple, after a firmware update, configurator stuck on "searching for controller", even if i plug and unplug my FC, need to restart the App to have the link
if i click on update FW, if I cancel the action, link is lost, need to quit app
If I want to restore a setup and I cancel this action, link is lost
hope it helps :)

noahwaldner, 2020-03-01:
what system are you on?
i assume windows?

make sure you don't have any other apps claiming the usb interface

ElMute, 2020-03-01:
I'm on Mac OS X 10,14,6

ElMute, 2020-03-01:
need to precise, no changes on my computer, and FC link worked great with configurator 1,04

noahwaldner, 2020-03-02:
Whats the order you use? 
plug in first or start software first?

I wanna try to reproduce the error here, i have a slight idea what could be the problem

PGIII, 2020-03-04:
Just Uploaded a new version of the configurator please try that and let me know if you still have the issue

ElMute, 2020-03-04:
yup, spoken with Noah in the beginning of the week and it's ok on my side

closed 2020-03-04 by PGIII