FlightOne/FalcoX-Issue-Tracker #3: Motors wont spin

FarFromMental, 2020-03-01:
Motors wont spin after clean firmware installation.

rayaan92, 2020-03-01:
I  am experiencing the same thing, after updating to the latest version my motors won't spin during setup.

gzaxr, 2020-03-01:
Same here. I had to calibrate them manual.

ESC: Megabolt Freestyle

rayaan92, 2020-03-01:
Tried to calibrate it manually to but everything else seems broken too. I
was unable to tune my rates nor disable sim mode

gzaxr, 2020-03-01:
> Tried to calibrate it manually to but everything else seems broken too. I was unable to tune my rates nor disable sim mode
> […](#)

Is there any way to downgrade? It seems impossible now

DaSeppoFPV, 2020-03-01:
same here, Motors wont spin after reflashing everything.
ESC: Mamba f50 pro
it works in blheli32, and if i flash betaflight to the board also everything works just fine....
i am also not able to flash an older Version via de configurator

PGIII, 2020-03-01:
Please try the latest version that was just posted and see if you have the issue still, should be fixed

bop4hell, 2020-03-03:
For me 10.0.6.2807 Beta fix it.

Thanks.

closed 2020-03-03 by PGIII