FlightOne/FalcoX-Issue-Tracker #5: Two motors stopped

SkeemoFPV, 2020-03-01:
Toothpick with Matek F4111 target. NamelessRC AIO412.
JESC 48khz with telemetry enabled
Set up went all fine, first test flight in auto-level was perfect for around 30s then two motors stopped. Tried to re-arm and still only two motors. Everything was ok after a power cycle, then again after about 30s two motors stopped.

PGIII, 2020-03-01:
Just Posted a new version, try that when you get a chance and let me know if you still have the issue

SkeemoFPV, 2020-03-02:
Unfortunately that has not fixed it, plus I now in get two motors spinning in the setup. I flashed backed to BF and confirmed everything was working, which it is, then DFU flashed the latest FalcoX, and still only have two motors working. I have tried both JESC 2.2 and 2.3 which made no difference.

SkeemoFPV, 2020-03-03:
The latest FW 10.0.6.2818 has fixed this problem for me, thank you, awesome work !!!!

closed 2020-03-03 by SkeemoFPV

ujenmr, 2020-03-19:
I still have problem on FW FLX_MATEKF411_10.0.6.2999.flx unfortunately falcox is unflyable for me