FlightOne/FalcoX-Issue-Tracker #7: Issues flashing Revolt V2

KrickeH, 2020-03-01:
**Version**
10.0.6.2804

**Target**
Revolt V2

**Describe the bug**
Can't load firmware on the board.

"Update to latest version" button loads Target: Bnf_mambaf411, Version: 1.0.0.2796 and Configurator just search for flight controller endlessly after that (Pic_1).

"Flash manually with file"  button (using FLX_REVOLT_10.0.6.2804.flx) stops on Loading blinking but never countinues to load the firmware ( waited over 15 minutes just to be sure) (Pic_2).

Flashing FLX_REVOLT_10.0.5.2527.flx manually works as it should (Pic_3)

**To Reproduce**
Happens everytime using normal firmware update procedures.

**Expected behavior**
Get Target: Revolt Version: 10.0.6.2804 working on the board :)

**Screenshots**

[image]
[image]
[image]

**Desktop**

- OS: Windows 10 pro 1909 18363.693
 - Browser: Microsoft Edge 80.0.361.62

PGIII, 2020-03-01:
That sounds like you have the wrong bootloader version, can you go into dfu mode and get the serial number of the board, has the board had its MCU replaced in the past ?

KrickeH, 2020-03-01:
Serial Number: 3085395F3235

And nothing is replaced / changed.

PGIII, 2020-03-01:
try updating through DFU mode now

KrickeH, 2020-03-01:
Working as it should now..

Excellent!

closed 2020-03-01 by KrickeH