FlightOne/FalcoX-Issue-Tracker #10: namelless aio f4 (matek f411 target) won't spin 2 motors

zeezyfpv, 2020-03-02:
I did boot update to new firmware 10.0.6.2804 (target matek f411) on my namelles aio f4 board.During the setup, 2 motors won't spin so couldn't finish setup succesfully.Also from the beggining I can't flash any previous *flx firmware only the latest through boot update.

SkeemoFPV, 2020-03-02:
This is the same issue as I am having please see issue #5

zeezyfpv, 2020-03-02:
@SkeemoFPV ok,tnx I've just seen your thread...anyway do you have problem flashing *flx firmware on your namelles board? I can only flash it with boot update so can't go back to firmware before

bop4hell, 2020-03-02:
Same with :
FC : Revolt OSD
ESC : AIKON-AK32-4IN1-V2
Motor : KO-2208-1900
During the setup, motors 3 and 4 "in Betaflight" won't spin...

Ok in FlightONE 1.2.94 Stable

SkeemoFPV, 2020-03-02:
> @SkeemoFPV ok,tnx I've just seen your thread...anyway do you have problem flashing *flx firmware on your namelles board? I can only flash it with boot update so can't go back to firmware before

Yes the same I can only use the boot button on the board to force into DFU, however this appears to be the same in BF

zeezyfpv, 2020-03-02:
@SkeemoFPV yes that's true...PG3 told me that that's something different in design with this board,but wasn't sure if I'm the only one that has that issue,I hope they will figure something up

PGIII, 2020-03-02:
Which two motors didnt spin, and did the motors that dont spin make the full start up tones or just the first beeps, also with the flashing try unplugging and plugging in the board to usb 4 times in a row it should go into bootloader mode, when the LEDs are flahsing fast thats bootloader mode

SkeemoFPV, 2020-03-02:
It’s the motors connected to M3 and M4 as marked on the board as far as I remember. I’ll check on the start up tone shortly, it’s quite hard to tell with such small motors.

SkeemoFPV, 2020-03-02:
> Which two motors didnt spin, and did the motors that dont spin make the full start up tones or just the first beeps, also with the flashing try unplugging and plugging in the board to usb 4 times in a row it should go into bootloader mode, when the LEDs are flahsing fast thats bootloader mode

It definitely seems to just be whats labelled M3 and M4, and they are all making the full start-up tones. I now only have two motors spinning in the wizard too.

bop4hell, 2020-03-03:
For me 10.0.6.2807 Beta fix it.

Thanks.

PGIII, 2020-03-03:
I just pushed a new version try that and let me know if it fixes the issue

zeezyfpv, 2020-03-03:
@PGIII It's all working fine on 2818,self level also..now the only thing remains not able to flash *flx files

PGIII, 2020-03-03:
> @PGIII It's all working fine on 2818,self level also..now the only thing remains not able to flash *flx files

Should open a separate issue for that, but that seems to be limited to only the Nameless board since the Tinyhawk I tested on works fine and uses the same target, does it go into boot loader mode if you unplug and plug in the usb 4 times in a row

zeezyfpv, 2020-03-03:
@PGIII Yeah I know, that nameless board has some different design,it doesn't go to bootloader mode with 4xusb thing,neither comunicate in terminal on commands,only with boot button.Same case in betaflight so that's not falcox related.But at least it's working fine with latest falcox.I'll open another thread for that,tnx for your effort

closed 2020-03-03 by PGIII