FlightOne/FalcoX-Issue-Tracker #11: Matek F411 - Motors don't spin on setup (running JESC 2.2)

Hellfire82, 2020-03-02:
10.0.6.2807 Alpha

Matekf411

Tinyhawk S which uses the Matek F411RX board, however I have installed an FrSky XM+ so am not using the SPI receiver.

FalcoX installs correctly & the receiver works, however when I reach the part of the setup wizard which asks if the motors are turning, none of them move even when the idle is set to 100%. Even when I bypass this by selecting yes, none of the motors turn in the next wizard to allow me to identify which of the 4 motors is turning. After bypassing this wizard by guessing which motors should be turning & power cycling, there is a constant flashing red light on the board.

In the wizard I selected BLHeli_S, however I am running JESC V 2.2 to take advantage of the 48kHz benefits.

PGIII, 2020-03-02:
Im pretty sure the MatekF411Rx is a different target completely so probably part of the issue

Hellfire82, 2020-03-02:
It is a different target, but the RX refers to the built in SPI receiver (which I'm not using).
You'll know more than me but I'm presuming my use of JESC instead of BlHeli_S may be the issue?

If this can't be rectified I may have to get a refund & go back to Betaflight :(

Hellfire82, 2020-03-03:
Flashed Matekf411 Version: 10.0.6.2818 Alpha and motors 2 and 3 now spin during the setup wizard, but motors 1 and 4 don't spin at all.

Flashed the 4in1 ESC back to BlHeli_S 16.7 and nothing changed, so I've reverted back to JESC 2.2.

Looks like this issue is fixable since 2 motors now spin due to what you released in Patch 3 for Beta 6 :)

PGIII, 2020-03-04:
[FLX_MATEKF411RX_10.0.6.2860.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/4289616/FLX_MATEKF411RX_10.0.6.2860.zip)
Try that firmware and see if it works, (Make sure to unzip it before you flash :) )

Hellfire82, 2020-03-04:
Getting closer!
All motors spin and I'm able to complete the setup wizard. However when I power cycle after the last step of setting up angle mode, the wizard starts again. It's as if it doesn't save.
Many thanks for going out of your way to find a solution. I know it's not an official supported target and you could've quite easily said no.
I'm FlightOne for life!

PGIII, 2020-03-04:
[FLX_MATEKF411RX_10.0.6.2863.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/4289867/FLX_MATEKF411RX_10.0.6.2863.zip)
This should fix the saving let me know

Hellfire82, 2020-03-04:
It's still not saving.
Flashed with both the old and the latest configurator (2.0.1) on three occasions to make sure.

Enlightx, 2020-03-11:
just tried it also getting the same run the setup wizard. motors all spin perfect... but then after you power cycle wizard starts again

Hellfire82, 2020-03-15:
Latest alpha 10.0.6.2999 has solved the saving issue and all motors spin.

closed 2020-03-15 by Hellfire82

Enlightx, 2020-03-19:
just tested this and i still only have 2 motors spinning on tinyhawk2 using stock ESC not jesc ?

gretel, 2020-05-03:
@PGIII does this support the builtin `spi` receiver?