FlightOne/FalcoX-Issue-Tracker #9: D value not well reported

ElMute, 2020-03-02:
**10.0.6.2807** but same issue with all version who're supporting DJI OSD
**Revolt Lite**

D well not reported on the DJI OSD, for exemple my D for pitch is 490, and I see 236 in the DJI OSD, seems D value is limit at 200 on the PID DJI OSD, because if i start at 150, and try to rise the D value, I can't exceed 200.

**To Reproduce**
just try to edit D value with a value better than 200, and go to DJI OSD to see what's value of D
is reported

PGIII, 2020-03-02:
Dji limits some fields, only solution would really be to make it 10 = 100, I’ll have to take a look and see

ElMute, 2020-03-02:
10 = 100 could be a good solution

xermalk, 2020-03-02:
Is this why i can navigate the pid values, but they keep resetting every half a second or so?

ElMute, 2020-03-02:
> Is this why i can navigate the pid values, but they keep resetting every half a second or so?

i Have this kind of issue when I tweak the filter frequency, but it seems it was when my lipo has low voltage, when my battery is well charged, i can adjust my filter with no resetting value issue

xermalk, 2020-03-02:
> > Is this why i can navigate the pid values, but they keep resetting every half a second or so?
> 
> i Have this kind of issue when I tweak the filter frequency, but it seems it was when my lipo has low voltage, when my battery is well charged, i can adjust my filter with no resetting value issue

I get this no matter the charge level of my goggle/quad battery.

https://www.youtube.com/watch?v=ZhPc5ek25LY

Coupled with how sketchy the pc osd redirection is, it was hell to get the quad flying.

Took half an hour to just get through the initial setup where i had to constantly reload the pc configurator because it only updated a fraction of the osd.

ElMute, 2020-03-02:
it's what i have when my quad lipo is on lower voltage, before flight I can update value from the DJI goggles to the FC with no problem, while in the end of flight, value jump like on your video. but perhaps it's not link to the battery level ...
since D value are not the right value, I don't use the PID section of the goggles, only the filters frequency sections, and sometimes it works flawlessly

ElMute, 2020-03-10:
hum, since the latest alpha, I can't update PID with the DJI OSD, value always rebooting to is first value

xermalk, 2020-03-12:
> hum, since the latest alpha, I can't update PID with the DJI OSD, value always rebooting to is first value

Thats exactly what i was experiencing all the time :)

closed 2021-06-10 by stale[bot]

Nilssonfpv, 2021-06-10:
Err, definitely not fixed and should not be closed :)