FlightOne/FalcoX-Issue-Tracker #12: namelles f4 aio board (matek f411 target) unable to flash *flx firmware

zeezyfpv, 2020-03-03:
Only this board specific issue due to it's different design,unable to flash *.flx firmware. The only way to flash it is with boot update to latest firmware in companion.Also this board goes to bootloader mode only with boot button.

noahwaldner, 2020-03-03:
This is probably a configurator issue i can look into. Is it stuck on loading? what exactly is happening?

zeezyfpv, 2020-03-03:
https://drive.google.com/file/d/1--S5vqqXKqkQsizd_1Bby9gNXdOzTSs_/view?usp=sharing
you can see in video,blue led on board start fast flashing,companion and terminal don't respond to commands at all like with other boards.This was in older companion,but with new version nothing changed. I also sent video before to @PGIII he knows about that problem.

noahwaldner, 2020-03-03:
The new configurator version will probably fix this.

zeezyfpv, 2020-11-06:
Hi,still the same problem on namelles matekf411taget aio board.I can only dfu boot to the latest firmware.No manual flashing flx files,board gets crazy.Tried also on latest companion 10.0.11,no change.Also with dshot8 which is the only protocol escs work with,motors sound like escs are skipping data,Before with dshot300 was fine.

closed 2021-06-10 by stale[bot]