FlightOne/FalcoX-Issue-Tracker #17: FalcoX 2823 AmpDraw numbers updating to quick

enrico320i, 2020-03-08:
**Version**
2823

**Target**
Revolt

**Describe the bug**
I'm using the Afterburner ESC with current wire Setup to MillivoltOSD v2. In FalcoX osd "AmpDraw" is showing the Number very fast updating so that it can not be reading anytime. Number is Jumping up and down crazy fast. It's not an Big issue, but mabey it can be fixed. The "mah used" feature is working fine. Also it does the same on an other Quad with Furling32 ESC.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Mabey it can be fixed in some feature Version. It Looks to me like it Needs Filtering like the Voltage Display in OSD, this works fine too. Seams the Amp Draw is getting no Filtering and gets updated live all the time.

**Screenshots**
Powerplay DVR file can be downloaded here from my Google Drive.
https://drive.google.com/file/d/1Pk9LF3oH8ITNFdiNq2Zps2yVo7bW1UGC/view?usp=sharing

**Additional context**
Add any other context about the problem here.

PGIII, 2020-03-10:
Weve been meaning to add a More filtered current draw reading for the OSD for a while ill have to take a look at it this week

enrico320i, 2020-03-10:
Thanks man, it's not a big issue and there is truly important stuff. But I
use it from time to time to see how high the amp draw is on a new setup 👌

Greetings from Germany Enrico

PGIII, 2020-12-17:
Latest version have more filtering on amp draw and should fix this issue

closed 2020-12-17 by PGIII