FlightOne/FalcoX-Issue-Tracker #18: Betafpv 1S F4 V2 NO RX Matek411 target issues

iggut, 2020-03-09:
**Version**
10.0.6.2823

**Target**
Matek411

**Describe the bug**
Using the NO RX version of the betafpv 1s f4 v2 board, the are multiple issues that could be related.
-During wizards upon requesting switch input I get channel 9 that appears as selection without any input on my side. This seems wrong as I cam using an XF nano in 8 chan mode.
-During flight the motors sound a bit off, as if they are giving more throttle than they should
-During flight and very often the drone just drops out of the sky disarmed and lands almost instantly armed without switch movements. Making it unflyable an making me think that whatever is sending the erroneous chan 9 signal is also sending the disarm signal for a split second.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash 2823 on betafpv 1s f4 v2 board with external XF nano and Tango 2
2. Try setting it up and flying
**Expected behavior**
Sexy fly time with FalcoXxX

**Additional context**
No issues with BF with same HW
ESC firmware tried: 
   -stock Blheli_S 16.6
   -JESC 2.3 24, 48, and 96khz
   -Jazz 16.7 24, 48, and 96khz
   -Jazz 16.8 RC3 24, 48 and 96khz

For the short times I do have control of the craft, it feels very robotic...love it
Let me know of any specific tests I can do to help you figure this issue out.

Thank you.

iggut, 2020-03-09:
Forgot to mention there is HUGE latency felt in all control inputs

iggut, 2020-03-09:
Tests done in DSHOT600 
When performed with DSHOT300, the latency is reduced by A LOT, but the drops out of the sky more often and when it does only motors 1 and 2 are able to re-arm.

PGIII, 2020-03-09:
What Uart is the RX on ?

iggut, 2020-03-09:
Uart 2

closed 2020-12-17 by PGIII