FlightOne/FalcoX-Issue-Tracker #19: FalcoX2823 Unify V3 5V not detected

enrico320i, 2020-03-09:
**Version**
2823

**Target**
Revolt

**Describe the bug**
Unify V3 5V not detected in VTX Wizard. I also can not select the VTX later in VTX page, it start searching and didn't found it. Works in FalcoX Beta 1751 normal under VTX type TBS Unify Pro HV. Smart Audio is connected to Tx3 as recommended, like i said works a Long time before.

enrico320i, 2020-03-10:
I test today to set the Uart manually to Smart Audio and see if I can select than.

enrico320i, 2020-03-10:
Ok it works, i can manually set UART3 to Smart Audio. After that i can select TBS Unify Pro HV and the Unify V3 5V is working fine. mabey for later Auto would be nice.

PGIII, 2020-03-13:
Are you using a Pit switch ?

enrico320i, 2020-03-13:
No Pit Switch in use.