FlightOne/FalcoX-Issue-Tracker #20: Numerous issues in Configurator 2.0.2

Hellfire82, 2020-03-11:
I am using the latest configurator V 2.0.2.
I am using the latest FalcoX alpha build - Revolt Version: 10.0.6.2823.

The following 4 issues have been identified so far:

[image]

When you click on the 'backup' button, the loading dialog box appears and nothing else happens (see screenshot above). I have waited for a considerable amount of time just in case of a delay before restarting the configurator.

[image]

When you paste / type commands into the serial terminal, nothing appears in the actual window like it used to e.g. in order to feedback info if a 'get' command is used or to say OK if a 'set' or 'save' command is used (see screenshot above).

[image]

When you click on the 'Configure OSD' button, the loading dialog box appears and nothing else happens (see screenshot above). I have waited for a considerable amount of time just in case of a delay before restarting the configurator.

[image]

When the configurator is hanging / not responding and either the reload or force reload option is selected within the 'view' tab, the screen just goes white and nothing happens (see screenshot above). The configurator has to be shut down and restarted in order for it to function again.

O/S - Windows 8.1 (all latest patches installed)
Browser - Version 80.0.3987.132 (Official Build) (64-bit)

closed 2020-05-06 by Hellfire82

gretel, 2020-07-05:
same issues for me. why closed?

Hellfire82, 2020-07-05:
All these issues have been fixed in the latest alpha configurator which is currently in testing.