FlightOne/FalcoX-Issue-Tracker #22: impossible to edit PID and Filter frequency with DJI OSD

ElMute, 2020-03-13:
**Version**
10.0.6.2823

Revolt Lite OSD

impossible to edit PID or Filter frequency with the DJI goggles, the value always reboot to his first value

go to PID with the DJI goggles, try to edit a value, you can see the value restart to the initial value

ElMute, 2020-04-06:
any news for this issue ? still there on the 3065

RoninMedia12, 2020-11-03:
Still occurring on Stable  4457 and previous builds not notes in changelog for fixing this in newer version 13.4523 either.

nes7901, 2020-11-12:
Setting up my first H7 with DJI. I'm getting OSD on googles but when I try to change pids and rates thru google values go right back to where they were at.  What is the proper setup for dji and be able to change pids, rates, and filter thru DJI google control?I I'm using the latest firmware 10.0.13.4523 Stable

dbrokenbone, 2020-11-21:
same here, cant change pid/rate

revolt lite on 10.0.15.4647
[image]

nes7901, 2020-11-24:
Im still having same issue

hajime-nohara, 2021-09-24:
I have same problem.
Will the problem be solved!?

dbrokenbone, 2021-09-24:
Nah... I changed it to Betaflight...