FlightOne/FalcoX-Issue-Tracker #23: Roll is reversed if you unplug battery and FC in certain Order

2dogrc, 2020-03-15:
**Version**
2992
"OK. Solved problem by myself. After firmware upload, do a power cycle, BUT... a BIG but.... battery must be first and USB cable after. IF USB is first and battery is connected after, then issue above. IF battery first and USB cable after then roll right is Right and roll left is Left 
In some old posts I mentioned few times that in some re-flashing or other troubleshooting issues - battery must be first."
Facebook user Alexandru Covaliov

PGIII, 2020-03-16:
What target is he using ?

2dogrc, 2020-03-16:
Revolt
https://www.facebook.com/photo.php?fbid=10215464186470255&set=gm.3111915955526852&type=3&eid=ARB8OEzsGAEq_7MvdzEfMq7dyJhd3jFVPty3BUZsKigU1M8Dg0M7jdv5tQbIT63hXum9PrX3L4VPFfwF&ifg=1

https://scontent-ort2-2.xx.fbcdn.net/v/t15.5256-10/s960x960/75550822_10215466069477329_7988109579617566720_n.jpg?_nc_cat=110&_nc_sid=1055be&_nc_ohc=vo9xRBDFQwIAX9tIRCu&_nc_ht=scontent-ort2-2.xx&_nc_tp=7&oh=a1b567a56e014cce2a80308b998d404c&oe=5E952726