FlightOne/FalcoX-Issue-Tracker #24: Randomly stops two motors on MATEKF411 and 10.0.6.2999

ujenmr, 2020-03-19:
**Version**
10.0.6.2999

**Target**
MATEKF411

**Describe the bug**
Every time randomly stops two (from right side) motors during fly

**Expected behavior**
flyable quad

**Additional context**
ESC firmware tried:
-Blheli_S 16.6 stock
-Jazz 16.77 48

ujenmr, 2020-03-19:
Last stable FW for me was FLX_MATEKF411_10.0.5.2527

SkeemoFPV, 2020-03-19:
I’m back to having only 2 motors working again as we had with an earlier release. 
Also I have the NamelessRC AIO412 and had been working on target MatekF411, on updating to the latest 2999 it’s now showing as MatekF4111RX which is for boards with the built in receiver, my board does not have a built in receiver.

PGIII, 2020-03-19:
> I’m back to having only 2 motors working again as we had with an earlier release.
> Also I have the NamelessRC AIO412 and had been working on target MatekF411, on updating to the latest 2999 it’s now [...]

Did you DFU flash ? there was an issue with it picking the wrong DFU bin it should be fixed now so please try updating in DFU mode and see if it solves the issue

SkeemoFPV, 2020-03-20:
> Did you DFU flash ? there was an issue with it picking the wrong DFU bin it should be fixed now so please try updating in DFU mode and see if it solves the issue

I always DFU flash as it’s really hit and miss getting it in bootloader. I even flashed BF first to be sure. 
I’ll test it straight away in the morning thanks for your quick reply.

ujenmr, 2020-03-20:
I use fc BetaFPV F4 2-4S AIO also with no rx (use crossfire), and configurator select MatekF4111RX and motors stops permanently. After manual selection of fw (https://flightone.com/firmware/alpha/FLX_MATEKF411_10.0.6.2999.flx) have earlier result - random stop 2 motors during fly

SkeemoFPV, 2020-03-20:
> Did you DFU flash ? there was an issue with it picking the wrong DFU bin it should be fixed now so please try updating in DFU mode and see if it solves the issue

@PGIII 
I have just DFU flashed and the MatekF411 is now correctly detected.
Unfortunately motors 2 and 3 (1 and 2 betaflight) only spin for a couple of seconds then stop.
I managed to get into bootloader mode and flashed #2818 and it worked perfectly, with no problems in flight at all.

ujenmr, 2020-03-20:
@SkeemoFPV Can you give me your fw, I don`t have flyable quad

PGIII, 2020-03-20:
> @PGIII
> I have just DFU flashed and the MatekF411 is now correctly detected.
> Unfortunately motors 2 and 3 (1 and 2 betaflight) only spin for a couple of seconds then stop.
> I managed to get into [...]

Try Dshot600 and see if you still have the issue

SkeemoFPV, 2020-03-20:
> Try Dshot600 and see if you still have the issue

@PGIII 
On quick test, it seems to be working with Dshot600

ujenmr, 2020-03-22:
For me solution - use the old version ((

PGIII, 2020-03-22:
> For me solution - use the old version ((

Did you try Dshot600 also what Rx are you using

ujenmr, 2020-03-23:
Motors spins only on dshot8 and multishot, rx - crossfire

ujenmr, 2020-03-25:
I just tested 10.0.6.2999 and thank you @PGIII, everything is ok on dshot600

closed 2020-04-01 by ujenmr