FlightOne/FalcoX-Issue-Tracker #25: CROSSFIRE TELEMETRY Not Working On any UART

Rafyel, 2020-03-19:
**Version**
Revolt Version: 10.0.6.2999 Alpha
Mambaf411 Version: 10.0.6.2999 Alpha
Matekf411 Version: 10.0.6.2999 Alpha

**Describe the bug**
Wired from any TBS unify or Tramp VTX to TX3 on the Revolt, Mamba, and Matek, and other available UARTS but telemetry doesn't work. It work with the older FlightOne firmware. Tried also on Mamba and Matex target as well, but there is no feedback to the transmitter.

**Additional context**
CROSSFIRE TELEMETRY DOESN'T WORK ON FalcoX

I used to get telemetry feedback on my radio with this connection, but now I have not for quite a few release. I'm using Spectrum iX12.

PGIII, 2020-03-19:
FalcoX Doesnt support telemetry yet,  it will be added in the future, but im not exactly sure what you mean because you mention Unifys and tramps so are you talking about VTX control or just telemetry to the Crossfire?

Rafyel, 2020-03-19:
I’m sorry, you’re correct. Just crsf telemetry. How soon will you implement it?

meekaah, 2021-01-12:
just a quick little bump on this to keep it alive 💯  i'd love to get that working

xermalk, 2021-01-24:
+1, with the dji system not storing the osd having the gps telemetry is *super* important.

vanawaker, 2021-02-05:
So I do not need to solder the RX3 pin to the crsf receiver right? Just TX1 no problem? Cuz no telemetry right now...