FlightOne/FalcoX-Issue-Tracker #27: PID value is set to 49 for the different profile

ElMute, 2020-04-06:
- would be great to could select our custom set up in the 3 profiles available, at this time PID are set to 49 for all parameters when we select "custom"

- would be great if the changes made could stay in place when we switch between the profiles, need to save after each edit before switch between the profiles or edit is lost during the switch

PGIII, 2020-04-07:
That just sounds like its not being set to defaults, we can fix that but try selecting the default preset and then edit a value and it will update the custom tune

ElMute, 2020-04-08:
yup it updates the custom in the profile selected, but for exemple : 

on my profile 2, if i want test only small adjustements of my custom preset from profile 1, I need to re-edit all the preset because all is set to 49. 

would be great if we could recall our custom preset from profile 1 to the 2 others profile

closed 2021-06-10 by stale[bot]