FlightOne/FalcoX-Issue-Tracker #33: Backup / Restore checkbox groups

Hellfire82, 2020-05-06:
On the Backup / Restore screen, the checkbox groups are not all visible unless you have the window maximised. There is no scroll bar to adjust this and they seem to be misaligned:

Windowed

[image]

Maximised

[image]

In addition it would be much easier if there was a 'check / uncheck all' button for these groups, if for example you only wanted to backup OSD.

 - OS: Windows 8.1
 - Browser: Chrome
 - Version: 81.0.4044.138 (Official Build) (64-bit)

PGIII, 2020-05-06:
What resolution is your monitor ?

Hellfire82, 2020-05-06:
1366x768 - The highest resolution my laptop will go to.

[image]

closed 2021-06-10 by stale[bot]