FlightOne/FalcoX-Issue-Tracker #37: New Configurator VTX Settings (Alpha Test)

SkeemoFPV, 2020-05-08:
In the VTX section the protocol and VTX type is just displayed as a number with nothing to reference too. 
Needs to either have a key or the actual names in the dropdown.
This is also the same for the Auto/PAL/NTSC box.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Configuration'
2. Click on 'VTX'
3. Observe the selection menus

**Expected behaviour**
I would like to see the actual name of the protocols and the cards listed.

**Desktop (please complete the following information):**
 - OS: macOS Mojave
 - Browser: Chrome
 - Configurator Version: Version 10.0.4-alpha
[image]

PGIII, 2020-05-19:
I pushed a new firmware a couple days ago that should make the VTX page work properly, and fixes some missing names

SkeemoFPV, 2020-05-19:
Yes that’s working great now thanks.

closed 2020-05-19 by SkeemoFPV