FlightOne/FalcoX-Issue-Tracker #40: no final 2 bip boot (alpha 3622)

ElMute, 2020-05-11:
no final bip boot with the 3622 version, can't use it.
tried to DFU flash, but it doesn't work correctly on the new configurator

PGIII, 2020-05-21:
Havent heard of anyone else having that issue what target, esc, and esc protocol are you using

ElMute, 2020-05-21:
set-up > revolt OSD lite, Megabolt Extreme, DShot 64 but same issue with DShot 32
i've 2 setup exactly the same, and i got this issue on both

PGIII, 2020-05-24:
Can you post a dump from the quad in here

ElMute, 2020-05-24:
> Can you post a dump from the quad in here

here it is 
[ElMute Tune ok.txt](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/4674617/ElMute.Tune.ok.txt)

ElMute, 2020-06-02:
> Can you post a dump from the quad in here

just to say, the final bips are here just after the flash, if the USB is plugged and connected to the configurator, as soon as i unplugged the FC, final bips are gone, and don't come back even if i replug the USB.
would be glad if you find a way to resolve this bug :)

PGIII, 2020-06-28:
3625 should resolve this issue

closed 2020-06-28 by PGIII