FlightOne/FalcoX-Issue-Tracker #46: New configurator Test - Cannot upgrade the FW to latest 3373

cross07705, 2020-05-13:
When plug in the FC first time, the new configurator detect the latest FW is 3373. But there is no response after clicking update. 
Then unplug and plug in the FC again, the new configurtor change the latest FW back to 3266, and update function works.
Is the new configurator only support 3266?
Thank you.

ElMute, 2020-05-13:
just flash one of my kwad with 3373 and old configurator, and this firmware don't work with the new alpha configurator. seems only 3266 can work with him

PGIII, 2020-05-18:
needs to be 10.0.7.x to support the new configurator, therell be a new update to the configurator soon that should fix it showing the wrong version number as the latest

ElMute, 2020-05-18:
great will try

cross07705, 2020-05-19:
Noted with thanks!

closed 2020-05-19 by cross07705