FlightOne/FalcoX-Issue-Tracker #50: Falcox Configurator 2.0.2 stuck at save/restore, configure OSD tabs

VictorWYVan, 2020-06-02:
**Describe the bug**
Companion is stuck showing 'Loading...' when I click 'Configure OSD' tab or 'Backup/Restore - backup'

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'FalcoX Configurator', plug in the quad
2. Click on 'Configure OSD' or 'Backup/Restore'
3. If clicked 'Backup/Restore', click 'backup'
4. See error

**Expected behavior**
Configurator hangs

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Enterprise
 - Browser chrome
 - Version Windows 10 1903

danbritt, 2020-06-08:
Also happening for me on MambaF411 target, same FW and configurator version as above.

I did notice that if you open the dev tools using the electron menu, you can see all of the commands being dumped to the console when you click the "backup" button. I can also see the commands being dumped to the console if I go to the terminal tab, type "dump", and then hit enter. So it appears the dump is working, it's just not presenting the user with an option to do anything with it.

pajalnic, 2020-07-27:
Also happening for me on  revoltOsd target, same FW and configurator version as above.

JRTrudel, 2020-09-25:
Very annoying bug...

PGIII, 2020-09-30:
This will be fixed in the upcoming release, the current prealpha configurator doesnt have this issue

closed 2020-10-04 by PGIII