FlightOne/FalcoX-Issue-Tracker #59: Buzzer always beep on

epcim, 2020-08-29:
**Version**
Revolt Connected, Version: 10.0.6.3923

**Target**
RevoltOSD 
(I did not spotted behaviour on on other quads with different buzzers)

**Describe the bug**
When battery is plugged in, beeper starts on after an timeout period.

**To Reproduce**
It was reproduced on two similar builds (2 persons), both with https://buzzer.hellgatefpv.com/product/hellgate-fpv-buzzer/

(three wire connection)

**Expected behavior**
- dont beep if battery is connected
- dont beep if battery is not connected but USB power

PGIII, 2020-09-30:
Where do you have it hooked up ? that buzzer may need to be attached to the pin toggle pin which is TX4 on the revolt, it looks for a high signal which is doesnt always get on the buzzer pad

PGIII, 2020-12-17:
Closing due to no activity

closed 2020-12-17 by PGIII