FlightOne/FalcoX-Issue-Tracker #61: [FEATURE] New Feature For Firmware

RoninMedia12, 2020-09-19:
**Is your feature request related to a problem? Please describe.**
Having to flash to BF to Use Blheli config. 

**Describe the solution you'd like**
Native Blheli32 pass through in FalcoX   Or ability to edit more setting in blheli32 in Falcox OSD:  motor timing, Demag, PWM, startup power.  

**Describe alternatives you've considered**
Old flight one method of having Cleanflight firmware flashable easily from the configurator.  

**Additional context**
Add any other context or screenshots about the feature request here.

PGIII, 2020-10-04:
This is already supported on Lightning

closed 2020-10-04 by PGIII

RoninMedia12, 2020-10-04:
So no support coming for revolt and revolt osd only new lighting board?

gretel, 2020-10-07:
why is this getting closed? customers asking for this feature. spending more money on another board which will not fit  smaller builds is not an option for me.

PGIII, 2020-10-20:
@gretel support for all targets was just added, also the lightning is smaller than the millivoltOSD so not sure how it wouldnt fit in a build

gretel, 2020-10-20:
@PGIII awesome, thank you. will there be an according release at https://flightone.com/download.php?version=alpha ?

RoninMedia12, 2020-10-20:
Thank very much PG3