FlightOne/FalcoX-Issue-Tracker #63: [FEATURE] DJI Air unit and Vista part of VTX config

RoninMedia12, 2020-10-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

DJI Integration setup procedure is not well outlined or intuitive .   have DJI option in telem and on uart page but not suppose to select dji also on FW 4457 DJI is not recognized on Uart 3 under telemetry.  and once I select DJI on Uart 3  it is locked and unable to change it to anything else.  

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Please Enable DJI air unit or Caddx vista as detection options for VTX config setup wizard and as an option on VTX menu, 
Once selected DJI just like other VTx's have sub-features to enable PID Control, OSD voltage Rssi, etc like you would have channel select and output power for analog VTX.  

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Full DJI OSD integration to support Falcox setup from DJI menu  

**Additional context**
Add any other context or screenshots about the feature request here.

PGIII, 2020-11-02:
> DJI Integration setup procedure is not well outlined or intuitive . have DJI option in telem and on uart page but not suppose to select dji also on FW 4457 DJI is not recognized on Uart 3 under telemetry. and once I select DJI on Uart 3 it is locked and unable to change it to anything else.

Detection should work, we can look into it if you're having issues, but there shouldn't be any need for setup outside of running the wizards, the UART page is mostly for debugging and will be hidden by default in the future.

>Please Enable DJI air unit or Caddx vista as detection options for VTX config setup wizard and as an option on VTX menu,
Once selected DJI just like other VTx's have sub-features to enable PID Control, OSD voltage Rssi, etc like you would have channel select and output power for analog VTX.

The DJI doesnt have channels that can be changed so it wouldnt make sense to be in the VTX menu, also theres not really any reason to turn off any of those features, theres not much to customize about DJI on the FC side. 

>Full DJI OSD integration to support Falcox setup from DJI menu

Waiting on DJI to allow full control over the screen :)

RoninMedia12, 2020-11-03:
Ok so on my one quad a wire was pinched I guess and was not fully connected replaced the RX wire now get osd so step one out of the way,  so it appears the auto-detect is working but if i go under the peripherals tab and do dji telemetry it still fails there for detection.   Also Pids and other items under tuning you can select and then change it but the value immediately changes back to what is was and only actually let you make any changes.

PGIII, 2020-12-01:
theres an issue open already about the PID turning issues which is being looked in to, did you try to detect it in the text menu after it was detected in setup or before ?