FlightOne/FalcoX-Issue-Tracker #64: Fl1 lightning f7 (Jhonn) no firmware installed

Ninobrow, 2020-10-31:
**Version**
nothing ,

**Target**
H7 lightning 

**Describe the bug*s
First connection all went good, going threw falco configurator until motor spin, sudently shot sown configurator.
Now when connecting battery or usb nothng happens, only video is steel working on vtx.
Falco x firmware = 0 and not possible to reflash by anyway.

**To Reproduce**
Steps to reproduce the behavior:
connect battery or usb and wait for falco to connect 

**Expected behavior**
Falco x configurator to open .

**Additional context**
I would need a back up file of the H7 if possible or any help to recover the FC

Thnk you all

PGIII, 2020-11-02:
please email support@flightone.com and they can help you reflash it, are the red leds blinking when you plug in the FC?

Ninobrow, 2020-11-06:
Hey,

Thanks for your response.

Yes they do, I finally managed to flash the latest version by entering in
bootmode (don't even know how I did) and saving the firmware on the FC file
and finally manual flashing.
Even tho, Falco does not recognize the Vtx (tbs32 hv) during set up, and
there is no way to get all the folders back that were initially on the FC
file (firmware/images/info/settings) I only have the flx file there...

No time to fly it yet.

I think I would definitely need a complete clean back up I guess and no one
around that I know with H7.

Will try my luck with support as advised.
Thanks
Best
D

closed 2020-11-17 by PGIII

Ninobrow, 2020-11-17:
Can you please raise a nee ticket ?
I have 3cards not working now (4 with the H7)