FlightOne/FalcoX-Issue-Tracker #65: DJI OSD Not properly showing Elements

RoninMedia12, 2020-11-14:
**Version**
 Lightning Version: 10.0.13.4523 Stable
 Revolt Version: 10.0.13.4523 Stable
Configurator 10.0.11

**Target**
Revoltosd
MilivoltOSD
H7 Lighting 

**Describe the bug*
Falcox OSD elements not showing properly in DJI custom osd
Only Voltage appears in the lower right default location next to the time in goggles white/red transition still appears to be working. 
I can not get any other FL1 osd elements to now show up, worked previously in either 3923 or 4457 was able to get arm status, pilot name and voltage also could move them around the screen.
I set up a second profile and moved all elements to the center of the screen as a test still would not appear in DJI OSD, but would show in the middle correctly in Configurators Home osd display.  
Unable to adjust Pids in DJI PID tuning Menu Values appears correctly but unable to modify. make change returns to the previous value.  also reported in Issue # 22 added note to that as well. 

**To Reproduce**
Steps to reproduce the behavior:
Attempt to use DJI OSD with current firmware. 

**Expected behavior**
Be able to use MSP accepted OSD elements with DJI goggles  would prefer current and comparison to what BF has,

Not sure if supported but like to see in OSD,  GPS,LQ, RSSI, Crosshair, 

**Screenshots**
Can provide phone pics if needed of DJI screen

RoninMedia12, 2020-12-02:
Update now do get custom elements in New firmware 4680 but Xfire and GPS still don't seem to be working correctly for me will update other GPS thread with notes on that.

PGIII, 2020-12-02:
> Update now do get custom elements in New firmware 4680 but Xfire and GPS still don't seem to be working correctly for me will update other GPS thread with notes on that.

When you say youre having issues with crossfire do you just mean the RSSI ?

RoninMedia12, 2020-12-02:
Yes Rssi/LQ for OSD is my issue Crossfire control has been working.   Good to know need to set LQ for Channel 8 no was unsure if this was different then BF or anything else and no docs on the wiki for setting it up. is this still video still mostly valid minus the fl1 gui part.    do need to set SF and output in Taranis/Opentx as well?  or just on the crossfire output map?

closed 2021-06-10 by stale[bot]