FlightOne/FalcoX-Issue-Tracker #69: Dji fpv rssi still not working with crsf it just shows 9

rosspeterson1, 2020-11-19:
**Version**
What version are you using, ex. 10.0.6.2804, can be found at boot up or in configurator

**Target**
What target are you using ex. Revolt

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

PGIII, 2020-12-01:
This is being looked into likely a conversion issue

RoninMedia12, 2020-12-02:
having same issue do we need to setup channel output on xfire rx at all?

PGIII, 2020-12-02:
> having same issue do we need to setup channel output on xfire rx at all?

Yes it needs to output LQ on channel 8 for the FC to see the data

closed 2021-06-10 by stale[bot]