FlightOne/FalcoX-Issue-Tracker #74: no gps detected

AirviewObscura, 2020-11-27:
in the OSD is no value for all the gps options shown

i´m using a beitian bn-22 glonas modul ...works great in the past.
havent updated millivolt for a while ..no i have installed latest update and gps is no moire working

using uart 6 for this ...in the past it works...
now on both free uarts ( 4 and 6) there is no gps detectetd when searching in menu for it.

is there any solution ? haven't change my setup the whole time

PGIII, 2020-12-01:
Theres a lot of people using GPS on the latest, what version did it work on last and what version are you using now ?

WilidWilly, 2020-12-01:
[FLX_REVOLT_GPS_DETECT_TEST.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/5626074/FLX_REVOLT_GPS_DETECT_TEST.zip)
Try this file out, I have a new detect gps and init setup

RoninMedia12, 2020-12-02:
I will test that file out and get back shortly but was having same issue in 4680 as well on h7  using Uart 8 for GPs and dji for uart 3   osd shows 14 sat in dji,  and in osd setup in config now have the GPS elements to choose from, BUT GPS dection wizard times out and does not detect the GPS so it seems like it partially recognizes it.   Also point to clarify to detect the GPS properly does the gps need Sats?  I am doing setup in my basement so no sats, I did fly the quad and never seemed to change and I ttl to usb and tested the GPS in Ublox it seemed to work just fine over baud rate 9600   is this also the correct baud rate for FL1 as it is not stated anywhere what the required baud rate is.    I am using a Beitian, BN 220 I got off Amazon.  I will test the private fix above and let you know here shortly.

RoninMedia12, 2020-12-02:
@WilidWilly  Just noticed this is for Revolt was about to move it to H7 firmware folder can you provide test build for Lighting or can I still use this one?

WilidWilly, 2020-12-02:
@RoninMedia12 ah you're right sorry let me get you the H7 version

WilidWilly, 2020-12-02:
@RoninMedia12 
[FLX_LIGHTNING_GPS_FIX.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/5632592/FLX_LIGHTNING_GPS_FIX.zip)

RoninMedia12, 2020-12-02:
@WilidWilly  flashed it but now osd will not show up in emulator in config so unable to test and board seem to just reboot and reconnect to windows every 30 secs now

WilidWilly, 2020-12-02:
@RoninMedia12 What UART is the gps on?

RoninMedia12, 2020-12-03:
Uart 8 as stated in my above comment.  I asked awhile back what uarts could be used never go a response I figure with H7 it should not really matter anymore.   I just did the av cable mod and the test firmware does not seem to be even fully booting I dont get all tones either, to even flashback to 4680 I had to delete the partition in windows disk management and have it auto reformat, as If I tried to do anything on the drive then after the initial flash of test fw it would just lock up the usb and reboot the fc.  But I am now back on 4680 and just did dection again and just times out and goes back one menu,  But the strange thing is now it will allow me to setup RTH on a switch as before would say no gps detected.  I went outside for a few minutes dam its cold and tried to get a lock but never did seem to  analog osd showed ) sats and Unlock symbol where the DJI osd was locked at 14 same as before.  I ensured GPS is set to Uart 8 in config.

PGIII, 2021-04-13:
This should be fixed in The latest Alpha

closed 2021-04-13 by PGIII