FlightOne/FalcoX-Issue-Tracker #76: Multiple Issues with Windows Configurator 10.0.12

RoninMedia12, 2020-11-27:
**Describe the bug**

Home Menu-
The main issue anytime FC requires a power cycle or other certain changes are made seems to never reconnect FC to the configurator unless you close the app and relaunch it.  Not a huge deal but does get very annoying having to relaunch the app multiple times.  if not just gets stuck at reading the FC FW version.  this occurs on f4 revolts targets.  have not tested on h7 as that boot sequence is different and have other issues with the configurator there. 

Configurator requiring Internet access...   this is very annoying in the field for tuning DJI and REvolt nonosd boards 
 
Other issues 
Home Menus-
1. setup drone button seems to do nothing if already have config, no way to reset config from configurator only osd?  
2. Backup and restore buttons do nothing. 
3. Test hardware button does nothing
4. replace the hardware button does nothing

Tuning menus
The pilot name still not able to be changed in Configurator just reverts back to fightone 

**To Reproduce**
Steps to reproduce the behavior:
connect f4 revolt to new configurator disconnect and reconnect,  

**Expected behavior**
Not having to re-launch app after fc disconnect or power cycle
backup, restore working properly 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] win 10 pro Version	10.0.19041 Build 19041
 - Browser [e.g. chrome, safari]  chrome
 - Version [e.g. 22]

**Additional context**
Android app would be nice and Dji tuning via goggles

PGIII, 2020-12-01:
The reconnect thing i have noticed, in general there are few connectivity issues were looking into, as far as the flashing the only flashing that would be slower would be DFU mode since its now following the standard more closely in terms of delay between packets, regular flashing through BL should be fast is that how youre flashing ? There are some buttons that are not implemented yet, internet access we can look into it shouldnt need it to work probably some unhandled error, also the pilot name isnt configurator related that should be fixed in the latest FW verisons

RoninMedia12, 2020-12-02:
For flashing I was doing both I think have so many fl1 fcs but even outside of flashing anytime for example you make a change to blheli protocol in the osd requires a Fc power cycle which in change requires you to also close the configurator or it wont reconnect properly, BF used to have similar issue before I remember,   I did find with the H7 it has different connectivity issues, it is very picky about when you connect it to the configurator or even open the configurator.  

With the H7 If I connect it to usb and already have the config open it seems to connect usually, and I can use stick inputs and enter osd but when  windows explorer finishes doing its drive discovery  and drive pops up it will not allow me to use my sticks at all in the config. seems to happen when on usb power only and quad powered on.   Now if I wait until folder is displayed close  that window and then open the config it will some times connect others it wont or get stuck in reconnecting loop.  Also the FC drive folder likes to pop up again like it reconnected to windows.  

I always get a windows error as well something went wrong and I figured that is normal do to how the drive is formatted or could this be part of the problem?

closed 2021-06-10 by stale[bot]