FlightOne/FalcoX-Issue-Tracker #85: Twitching Motors on stable release

PaceFPV, 2020-12-28:
**Version**
- All Versions above 10.0.6.3923 beta-release are going to give the same behavior

**Target**
- F405 based problem. RevoltOSD, MillivoltOSD

**Describe the bug**
- Glitch/Twitching motors on startup.

**To Reproduce**
- update from 10.0.6.3923 to any stable release AND!! reset the fc configuration.

**Additional context**
https://www.facebook.com/groups/falcox/permalink/1157647701319434

Everytime i update any F405 based fl1 fc with the latest stable release i get twitching motors. It flew like a charm with no problems on defaults and agressive tunes (on latest beta).
Buildquality is very clean and hardware is installed as recommended.

- MCU facing upwards
- cables routed outside the stack
- installed 470uf cap on vbat
- recommended esc settings (26 degree, 48khz)
- no stack nuts on top!

PGIII, 2020-12-28:
What Rx, tune, and ESC protocol are you using ? and what Tune and ESC protocol were you using on the beta version

PaceFPV, 2020-12-28:
RX is crossfire nano on 150Hz mode. ESC protocol dshot32, but was also present on a different setup with blHeliS esc and dshot8.
I just used default tune on both (beta and stable).

PGIII, 2020-12-29:
Are you sure its only after resetting the config ? Can you replicate that on the current stable version ?

PGIII, 2021-01-15:
Try that latest alpha pretty sure it should resolve this issue