FlightOne/FalcoX-Issue-Tracker #86: LED control broken

xermalk, 2020-12-28:
**Version**
Alpha: 10.1.0.4808 AND Stable: 10.1.0.4807

**Target**
HEADSUP LIGHTNING

**Describe the bug**
LED control is broken on both the regular led out pad, and the RX5/alternate led pad.

**To Reproduce**

1. Connect leds
2. Set led number and color (and alternate RX5 led pad)
3. Power on quad

**Expected behavior**
All 20 leds set to green 50%.

**Actual behavior** 
No leds lit until its first powered up over usb, and then battery connected.
Actual leds that light up, Random colors/unlit leds.

STABLE
The Matek slim ledstrip flash random patterns of green. see video
Using the LED pad i get 19 lit leds with it set to 20. with the 19th led slightly off color.

ALPHA
Using Alternate/RX5 in the alpha. I get (solid) random colors and unlit leds across all 20 on the racewires.

**Screenshots**
[image]

**Video**
https://www.youtube.com/watch?v=frTzGlE3vdQ

**Additional context**
Tried both the stable with the regular led pad, and the alpha with the alternate led pad after talking to Nick.
Both strips work just fine if used on a Matek F405 wing fc running inav.

PGIII, 2020-12-28:
Which pad is it attached to in the video ?

xermalk, 2020-12-28:
> Which pad is it attached to in the video ?

In the video the WS2811 8 led strip is connected to the LED pad.
Ill route a few more of my wires, so that i have the matek 2811 leds also connected to the RX5 pad.

The iFlight strip is using the silly bright 3535 leds. aka [SK6812MINI](https://cdn-shop.adafruit.com/product-files/2686/SK6812MINI_REV.01-1-2.pdf)

xermalk, 2020-12-28:
This is a video of the RX5 output, and with a
10 -- 10
 |-8
wiring of the leds (aka one arm parallell, that then continues to the other arm).

https://www.youtube.com/watch?v=LvIVc2uOVzY
This is probably not quite the expected behavior. :D

Bottom leds and second arm top leds don't light up at all. and more and more leds activate whenever i arm and disarm the quad.

To bad i don't have a oscilloscope as i guess you'd really need to see the actual output signal to get any useful data.

PaceFPV, 2020-12-28:
@PGIII i can confirm that the problem is also present on f405 millivoltOSD

Fettec Racewire LED connected to LED pin

PGIII, 2020-12-29:
Do the LEDs on the Millivolt act any different with the beta firmware ? [https://flightone.com/download.php?version=beta](url)

Enlightx, 2021-01-24:
Following as i have the same. led just random do what they want when they want

closed 2021-06-10 by stale[bot]