FlightOne/FalcoX-Issue-Tracker #89: [FEATURE] Motor test/rampup for one motor at a time

xermalk, 2020-12-31:
**Is your feature request related to a problem? Please describe.**
Testing for vibrations/bad bearings

**Describe the solution you'd like**
A way to spin up and test one motor at a time, just like the motors tab in clean/betaflight. 
Bonus points for actual vibration stats in the osd, but not really needed. You will hear a bad bearing once you start to ramp up the throttle.

It is however quite hard to hear what motor it is when all are spinning. 

**Additional context**
[image]

krunked-fpv, 2020-12-31:
Use blheli.....

PGIII, 2020-12-31:
Motor test is already slated for 10.1, one thats similar to what weve had previously, will likely be available for the public to test in one of the next few alphas

xermalk, 2020-12-31:
> Use blheli.....

Actually doesn't work ;) blhelisuit uses the betaflight code when your in that tab.

Great that its being worked on PGIII.