FlightOne/FalcoX-Issue-Tracker #91: REVOLT OSD - Crossfire 50hz failsafe

loutwice, 2021-01-10:
**Version**
FLX_REVOLT_10.0.18.4939 

**Target**
RevoltOSD
tango 2 pcb v1 

**Describe the bug**
when crossfire is set at 50hz - it failsafe 

**To Reproduce**
Steps to reproduce the behavior:
set crossfire to 50 hz - 
arm the kwad play with the throttle you will already feel a delay 
and it failsafe

**Expected behavior**
not failsafe 

**Additional context**
Beta firmware FLX_REVOLT_10.0.6.3923 works really well no problem at all

PGIII, 2021-01-11:
Do you still have this issue on the latest alpha version ?

loutwice, 2021-01-11:
> Do you still have this issue on the latest alpha version ?

yes and it comes even faster than before

loutwice, 2021-01-13:
let me know if you want special test or anything else

PGIII, 2021-01-17:
[FLX_REVOLT_10.1.0.5160_SERIAL_DEBUG_INFO.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/5827025/FLX_REVOLT_10.1.0.5160_SERIAL_DEBUG_INFO.zip)
Can you get a DVR with this

loutwice, 2021-01-21:
> [FLX_REVOLT_10.1.0.5160_SERIAL_DEBUG_INFO.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/5827025/FLX_REVOLT_10.1.0.5160_SERIAL_DEBUG_INFO.zip)
> Can you get a DVR with this

back from holydays ! sorry for the delay ! here is the video let me know if you need more 

https://user-images.githubusercontent.com/7931460/105342678-c1897d00-5c34-11eb-8e08-7c98ec6893fb.mp4

PGIII, 2021-01-23:
Hmm thats pretty Strange, doesnt look like a Serial Port issue, ill have to give you another Test bin, do you have the VTX attached to the Receiver or anything like that ? and how come you use 50hz do the faster speeds not have an issue ?

loutwice, 2021-01-23:
> Hmm thats pretty Strange, doesnt look like a Serial Port issue, ill have to give you another Test bin, do you have the VTX attached to the Receiver or anything like that ? and how come you use 50hz do the faster speeds not have an issue ?

Yes I use tbs vtx linked to the tbs nano so I can change power and channel directly in the crossfire menu from the remote - I will have to retry with 150 hz to confirm you if it works - 

Yesterday for a work I flew 10.0.6.2823 - no problem with that old one. If you want I can try all firmware that went out from this one so we can find from when it is happening - just throw me the firmwares you want me to test -

loutwice, 2021-01-29:
forget to mention it happens also with without vtx on nano rx ( got an hd kwad behave the same ) i will try it soon on H7 see if it happens

PGIII, 2021-03-16:
Closing since this seems to be caused by the same problem as #144

closed 2021-03-16 by PGIII