FlightOne/FalcoX-Issue-Tracker #94: can't finish motor setup - motor won't spin

svoland, 2021-01-12:
**Version**
10.1.0.5043

**Target**
omnibusf4sd - lanrc f4 v3s

**Describe the bug**
motor setup wont end - motor wont spin at any idle %

**To Reproduce**
Steps to reproduce the behavior:
1. start setup proccess
2. go to motor spin test

**Expected behavior**
the motor should start spinning at some idle %

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

PGIII, 2021-01-13:
I cant find much info on the board its for sure a omnibusf4sd target ? also what esc are you using ?

svoland, 2021-01-15:
> I cant find much info on the board its for sure a omnibusf4sd target ? also what esc are you using ? 

I used lanrc stack, link below. It's came with this target and works with BF fine

https://www.joom.com/en/products/5fbcc3768248510107c8c3e7

PGIII, 2021-01-15:
Try 10.1.0.5135 should fix issues with blheli_s escs not spinning

svoland, 2021-01-15:
> Try 10.1.0.5135 should fix issues with blheli_s escs not spinning

Fixed

closed 2021-01-15 by svoland