FlightOne/FalcoX-Issue-Tracker #95: won't find iflight VTX

svoland, 2021-01-12:
**Version**
10.1.0.5043

**Target**
omnibusf4sd - lanrc f4 v3s
VTX - iFlight succex force - uses irc trump protocol

**Describe the bug**
on the initial setup, wont find any vtx connected ( the same setup works with betaflight )

**To Reproduce**
Steps to reproduce the behavior:
1. start fc config 
2. wont fount any vtx

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

PGIII, 2021-01-15:
Try 10.1.0.5135 and see if you still have the issue

svoland, 2021-01-15:
> Try 10.1.0.5135 and see if you still have the issue

Still have the issue. 
VTX connected to tx3.

svoland, 2021-01-17:
The manual selection works fine for SA and irc Tramp. The auto detect won't work for both, for different TX on the board

PGIII, 2021-01-19:
What do you mean Manual works for SA and Tramp, doesnt that VTX only use the tramp protocol ?

svoland, 2021-01-19:
> What do you mean Manual works for SA and Tramp, doesnt that VTX only use the tramp protocol ? 

Sorry,
Tested it with iFlight VTX ( irc Tramp) and with rushtank VTX ( SA 2.0 ). On 2 different uarts. Same results - auto detect didn't worked for both. Manual configuration worked well for both

closed 2021-06-10 by stale[bot]