FlightOne/FalcoX-Issue-Tracker #97: FC loses connection with RX on the first power cycle after setup.

Tonsuke201450, 2021-01-14:
**Version**
10.1.0.5084

**Target**
Omnibusf4sd

**Describe the bug**
FC loses connection with RX (Frsky XM+) on first power cycle after the completion of initial setup wizard.

**To Reproduce**
1.  Flash Omnibusf4sd target with XM+ connected and bound to TX.
2.  Complete initial setup wizard with no problem.  That means RX is properly connected with both FC and TX during setup.
3.  Power cycle FC as instructed.
4.  FC cannot communicate with RX any more and impossible to do anything.

**Expected behavior**
Should be able to go into OSD menu and assign switches, change settings and fly.

**Additional context**
RX is connected to UART6 with inverted SBUS.  It is automatically detected by initial setup.

PGIII, 2021-01-14:
can you post a dump from the quad

Tonsuke201450, 2021-01-15:
Here you go.
[FalcoX_TNSK2502HD_OMNIBUSF4SD_cli_20210115_102919.txt](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/5817840/FalcoX_TNSK2502HD_OMNIBUSF4SD_cli_20210115_102919.txt)

PGIII, 2021-01-15:
So are when you move the arm switch does the arm status move at all ? can you go through setup and get a dump when the setup is done before you reboot, i also just pushed a new version so try that as well and see if anything is different

Tonsuke201450, 2021-01-16:
Arm status not changed at all after power cycle.  I updated the firmware for 5142 but it did not solve the issue.  Attached please find a dump taken right after the completion of setup wizard, before reboot.  Thanks for your attention.
[FalcoX_TNSK2502HD_OMNIBUSF4SD_cli_202101165_100419.txt](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/5823577/FalcoX_TNSK2502HD_OMNIBUSF4SD_cli_202101165_100419.txt)

PGIII, 2021-01-19:
[FLX_OMNIBUSF4SD_10.1.0.5176-RCTEST.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/5837901/FLX_OMNIBUSF4SD_10.1.0.5176-RCTEST.zip)
Can you try this and get me and see if the numbers move with the sticks

Tonsuke201450, 2021-01-20:
I tried but found no fix.  It is showing NE/FE number counting up but only zeros for axis and throttle.  I reset config and went through wizard successfully but nothing changed after the power cycle.  I also tried 5183 but nothing changed.
[image]

closed 2021-06-10 by stale[bot]