FlightOne/FalcoX-Issue-Tracker #99: OSD for Mamba stack F405 MKII (Fury F4 OSD) did not work

Skapix65, 2021-01-19:
Hi!
I'have just try to instal FalcoX on my Mamba stack F405 MKII (Furyf4osd)
Flight Controller Version: 10.1.0.5164
App Version: 10.1.2-alpha

Every thing works but i can't see the OSD in my google (i can see it in the configurator)
i have try to change NTSC or PAL in my camera and in the configurator (and power cycle) but the OSD still not working in my google
My VTX is detected (Tramp HV)
I'have try on Mac and PC but it's the same result

PGIII, 2021-01-19:
If its a FuryF4OSD Target the MambaF411 target isnt going to work on it

Skapix65, 2021-01-19:
> If its a FuryF4OSD Target the MambaF411 target isnt going to work on it

[Photo of the problem](https://www.facebook.com/photo/?fbid=10225755277450266&set=gm.3972409216144184&__cft__[0]=AZUR_F0fTtOcN3sPjI0j-n0jewhazxlNF7B05ON7f5Q7dLHGIVOZghzihWQyT1slOnTx1zqiNPnkmfY6H3zRdGWfPs6ihq0s9lv7a0Jm4OXrBWeHlFGVzMhg09hyrJVIyeazqcXywyQ6j2Lzvk_RVgxbtqSoB89gI97AsMB3Fd-F4A&__tn__=EH-R)

FuryF4 OSD Target is instal on my MambaF405, i had to buy f411 license just to unlock the bootloader

I think it may just be configured wrong for analog in the test target because it seems to work for MambaF405 DJI

Xai5er, 2021-01-19:
I have the same Problem with the FuryF4OSD Target. see the picture: https://ibb.co/WW0qgq0

PGIII, 2021-01-19:
> > If its a FuryF4OSD Target the MambaF411 target isnt going to work on it
> 
> [Photo of the problem](https://www.facebook.com/photo/?fbid=10225755277450266&set=gm.3972409216144184&__cft__%5B0%5D=AZUR_F0fTtOcN3sPjI0j-n0jewhazxlNF7B05ON7f5Q7dLHGIVOZghzihWQyT1slOnTx1zqiNPnkmfY6H3zRdGWfPs6ihq0s9lv7a0Jm4OXrBWeHlFGVzMhg09hyrJVIyeazqcXywyQ6j2Lzvk_RVgxbtqSoB89gI97AsMB3Fd-F4A&__tn__=EH-R)
> 
> FuryF4 OSD Target is instal on my MambaF405, i had to buy f411 license just to unlock the bootloader
> 
> I think it may just be configured wrong for analog in the test target because it seems to work for MambaF405 DJI

The F411 Bootloader wont work for an F405 target, what is youre serial number so i can swap the type, also are you plugging the battery and then USB the OSD is usually only powered by battery on the mamba boards

Skapix65, 2021-01-19:
> The F411 Bootloader wont work for an F405 target, what is youre serial number so i can swap the type, also are you plugging the battery and then USB the OSD is usually only powered by battery on the mamba boards

Here is my serial number : 206D3477534D
Thanks for the swap!

Ok i will powered the OSD with the lipo

Skapix65, 2021-01-20:
> The F411 Bootloader wont work for an F405 target, what is youre serial number so i can swap the type, also are you plugging the battery and then USB the OSD is usually only powered by battery on the mamba boards

Can you explain me what did i do wrong? Maybe i just have to tel us i have a Diatone Mamba F405 in the Order notes with the serial number?

I have just try to update bootloader but i still have F411 Bootloader...

martukas0210, 2021-01-20:
Same problem with my MambaF405. I used MatekF411 code. Can 

> The F411 Bootloader wont work for an F405 target, what is youre serial number so i can swap the type, also are you plugging the battery and then USB the OSD is usually only powered by battery on the [...]

Same problem with my MambaF405. I used MatekF411. Can u swap type too? My  serial No.: 375535563339

Skapix65, 2021-01-20:
I've just try another time, someone swap the target but you have make a mistake 

Now it's MambaF405US and i need FuryF4OSD! 

Thanks

PGIII, 2021-01-20:
> Can you explain me what did i do wrong? Maybe i just have to tel us i have a Diatone Mamba F405 in the Order notes with the serial number?
>
> I have just try to update bootloader but i still have F411 [...]

Many Targets are Being Added/Tested so theyre are not on the site right now, there will be a way to purchase a license code soon which will let you activate a FC through the Configurator, i updated those serial numbers they should both be furyf4osd targets now

Skapix65, 2021-01-20:
> Many Targets are Being Added/Tested so theyre are not on the site right now, there will be a way to purchase a license code soon which will let you activate a FC through the Configurator, i updated those serial numbers they should both be furyf4osd targets now

Thank you very much! Target is ok on my FC (Fury F4 OSD)
BUT i still didn't have OSD... :(

Did i have to make something with the sérial?
My serial 
UART1 (Tramp télém)
UART2 (OFF)
UART3 (CRSF)

I realy don't know what do for this OSD...

martukas0210, 2021-01-20:
Everything works except: on UART3 crossfire not working but works on UART1, on UART6 I didn't try. No OSD, as Skapix, (on BF works, Ichecked to be assured).

Skapix65, 2021-01-22:
Can someone tell us if someone is working on a new Flight Controller Version for this Mamba stack F405 MKII (Furyf4osd) ?
Because I don’t think anyone made this FC work good on FalcoX. Everyone who try it have the same OSD problem... One guy make it work with the DJI OSD (only on UART 6) but he have realy bad oscillations in flight

PGIII, 2021-01-22:
> Can someone tell us if someone is working on a new Flight Controller Version for this Mamba stack F405 MKII (Furyf4osd) ?
> Because I don’t think anyone made this FC work good on FalcoX. Everyone who [...]

Many of the targets on the alpha page are basically prealpha as they are ones I don’t have here and are being tested so I don’t expect them right away, I’ll have to check the OSD settings for this target there may just be something not set right, but to verify you’re plugging in the battery first when testing the osd right?

Skapix65, 2021-01-22:
> Many of the targets on the alpha page are basically prealpha as they are ones I don’t have here and are being tested so I don’t expect them right away, I’ll have to check the OSD settings for this [...]

No problem i understand! :)
Thanks a lot for helping us!
Yes... I have tried a lot of things (plugging before, during and after, on mac, on pc, whith different UART...)
I have made all the update and setup procédure with the osd of the configurator (with lipo plugged)
Every thing work without the OSD (don't even try to fly but arming is ok)

patrush, 2021-01-22:
Got same problem here no osd appears on googles

PGIII, 2021-01-22:
> No problem i understand! :)
> Thanks a lot for helping us!
> Yes... I have tried a lot of things (plugging before, during and after, on mac, on pc, whith different UART...)
> I have made all the update [...]

Ok I think I do see a problem with the target settings there will probably be a new version to try today

Skapix65, 2021-01-22:
> Ok I think I do see a problem with the target settings there will probably be a new version to try today

Thank you so much for your responsiveness! I will try every 10mn ^^
Can you tell us when it's ok?
I will give you a feedback if it's work for me

PGIII, 2021-01-23:
> > Ok I think I do see a problem with the target settings there will probably be a new version to try today
> 
> Thank you so much for your responsiveness! I will try every 10mn ^^
> Can you tell us when it's ok?
> I will give you a feedback if it's work for me

Just Uploaded a new Version on the alpha page try that and let me know if it fixes it

Skapix65, 2021-01-23:
> Just Uploaded a new Version on the alpha page try that and let me know if it fixes it

Good job mate! OSD is working for me!
I will try to flight tomorow,
Thank you so much!

PGIII, 2021-01-24:
No problem :)

closed 2021-01-24 by PGIII