FlightOne/FalcoX-Issue-Tracker #100: Buzzer won't work on Hobbywing Xrotor (OMNIBUSF4SD) FC.

Tonsuke201450, 2021-01-20:
**Version**
10.1.0.5183

**Target**
OMNIBUSF4SD (Hobbywing Xrotor F4 G3 FC)

**Describe the bug**
Buzzer won't beep despite it is correctly installed hardware wise and setup with OSD.

**To Reproduce**
1. Solder a buzzer accordingly.
2. Setup buzzer switch in OSD.
3. No beep.

I am experiencing buzzer never stops with MatekF411 target as reported in #30 https://github.com/FlightOne/FalcoX-Issue-Tracker/issues/30  Seems FalcoX for 3rd party edition tend to have issues with peripherals.

PGIII, 2021-01-20:
What kind of buzzer are you using ?

Tonsuke201450, 2021-01-24:
It was found that buzzer doesn't work with Mamba F405 Mk2 (FURYF4OSD) as well.  Seems something wrong with peripheral handlings of FalcoX.

PGIII, 2021-01-24:
> It was found that buzzer doesn't work with Mamba F405 Mk2 (FURYF4OSD) as well. Seems something wrong with peripheral handlings of FalcoX.

Theyre new targets that i dont have here to test im sure not everything will work perfectly the first time, but open a seperate issue for the Fury target ill take a look at them this week

Tonsuke201450, 2021-01-24:
Sure.

PGIII, 2021-01-29:
Try Version 10.1.0.5252 |10.1 Alpha 15| and see if you still have the issue

Tonsuke201450, 2021-01-29:
I confirmed it is working !  Thanks PGIII !

closed 2021-01-30 by Tonsuke201450