FlightOne/FalcoX-Issue-Tracker #103: Buzzer won't work on Mamba F405 Mk2 (FURYF4OSD) FC.

Tonsuke201450, 2021-01-24:
Version
10.1.0.5196

Target
FURYF4OSD (Mamba F405 Mk2 FC)

Describe the bug
Buzzer won't beep despite it is correctly installed hardware wise and setup with OSD.

To Reproduce

Solder a buzzer accordingly.
Setup buzzer switch in OSD.
No beep.

dustyprops, 2021-01-27:
I'm facing the same issue. Everything wired correctly (Buz+ & Buz-). 2 wire buzzer does not beep.

PGIII, 2021-01-29:
Try Version 10.1.0.5252 |10.1 Alpha 15| and see if you still have the issue

dustyprops, 2021-01-29:
> Try Version 10.1.0.5252 |10.1 Alpha 15| and see if you still have the issue

Okay will do. Thanks!

dustyprops, 2021-01-29:
> Try Version 10.1.0.5252 |10.1 Alpha 15| and see if you still have the issue

Buzzer working now with alpha .5252, great job!

Tonsuke201450, 2021-01-30:
I confirmed it's fixed.  Thanks for your work !

closed 2021-01-30 by Tonsuke201450