FlightOne/FalcoX-Issue-Tracker #106: [FEATURE] New Feature For Firmware - BLADE_F7

V1NCEFPV, 2021-01-26:
Hi,

Can you please add BLADE_F7 in Alpha Firmware ?

Thanks a lot guys.

[BTFL_cli_HolyShiter_20201224_001227.txt](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/5874849/BTFL_cli_HolyShiter_20201224_001227.txt)

WilidWilly, 2021-02-01:
It is already in the alpha firmware now :)

V1NCEFPV, 2021-02-01:
Yeah 🙂I know, already installed and ready to test in my Vega Gravity 😁🤙Envoyé depuis mon appareil Galaxy
-------- Message d'origine --------De : WildWillyFPV <notifications@github.com> Date : 01/02/2021  01:42  (GMT+01:00) À : FlightOne/FalcoX-Issue-Tracker <FalcoX-Issue-Tracker@noreply.github.com> Cc : V1NCEFPV <flash2999@gmail.com>, Author <author@noreply.github.com> Objet : Re: [FlightOne/FalcoX-Issue-Tracker] [FEATURE] New Feature For Firmware - BLADE_F7 (#106) 
It is already in the alpha firmware now :)

PGIII, 2021-03-15:
now supported

closed 2021-03-15 by PGIII