FlightOne/FalcoX-Issue-Tracker #108: [FEATURE] New Feature For Firmware TMOTORF7

V1NCEFPV, 2021-01-26:
Hi,

Another alpha firmware request for TMOTORF7.

Thanks a lot.

[033_EMUF_cli_DEADPOOL_20210114_154031.txt](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/5874860/033_EMUF_cli_DEADPOOL_20210114_154031.txt)

Vitrifly, 2021-02-01:
Has the firmware been released yet?

V1NCEFPV, 2021-02-01:
TMOTORF7, no not yet. Envoyé depuis mon appareil Galaxy
-------- Message d'origine --------De : Vitrifly <notifications@github.com> Date : 01/02/2021  20:03  (GMT+01:00) À : FlightOne/FalcoX-Issue-Tracker <FalcoX-Issue-Tracker@noreply.github.com> Cc : V1NCEFPV <flash2999@gmail.com>, Author <author@noreply.github.com> Objet : Re: [FlightOne/FalcoX-Issue-Tracker] [FEATURE] New Feature For Firmware TMOTORF7 (#108) 
Has the firmware been released yet?

V1NCEFPV, 2021-02-01:
You can check there : https://flightone.com/download.php?version=alpha😉🤙Envoyé depuis mon appareil Galaxy
-------- Message d'origine --------De : Vitrifly <notifications@github.com> Date : 01/02/2021  22:46  (GMT+01:00) À : FlightOne/FalcoX-Issue-Tracker <FalcoX-Issue-Tracker@noreply.github.com> Cc : V1NCEFPV <flash2999@gmail.com>, Author <author@noreply.github.com> Objet : Re: [FlightOne/FalcoX-Issue-Tracker] [FEATURE] New Feature For Firmware TMOTORF7 (#108) 

Thanks!

PGIII, 2021-03-15:
10.1 alpha 16 adds support for this target

closed 2021-03-15 by PGIII