FlightOne/FalcoX-Issue-Tracker #116: No FC startup tones when battery plugged. Motors don't spin in initial setup IFF722 TWING

kennytakfpv, 2021-01-28:
**Version**
10.1.0.5252

**Target**
IFlight F722 TwinG

**Describe the bug**
No FC tone when battery is plugged. Motors don't spin in initial setup and FC orientation setup can't be achieved. No data from accelerometer in setup.

**To Reproduce**
Launch Alpha configurator, wait for osd to start. Run initial setup and motors don't spin when asked if they do.
Click yes and skip motor setup, setup fail to get data frome accelerometer when aked if quad is laid flat and get any further.

**Expected behavior**
FC tones to be heard after battery is plugged. Motor start to spin in setup. FC orientation setup to work.

**Screenshots**
Not relevant

**Additional context**
None

marcelinojorgeromero, 2021-01-30:
I confirm the error. Motors don't spin at all and the ESC does not complete the last two step tone.

mbowkm, 2021-01-30:
I have exactly the same issue. Can't get the motors to spin up in the setup (they worked perfectly fine on the BF install I had on the FC previously, so it's not a mechanical or wiring issue)

marcelinojorgeromero, 2021-02-01:
I tried to install the firmware on a different quad with the same stack on it but different components (CSRF, Air Unit, GPS BT220, Leds) and the ESC was able to complete the tones. During the setup i couldn´t get the motors to spin but i skipped all the steps until the end.
The FC and ESC are sideways (maybe 90 degrees) and i had configured a custom sound from the BLHeli32 configurator. In the end when i armed the quad 3 out of 4 motors were spinning except the FR motor and BLHeli32 could read the setup successfully.

[image]

jokker87, 2021-04-20:
+1 as in #169

HrrrPippi, 2021-06-02:
Did you get this resolved? I have the exact same problem

PGIII, 2021-08-14:
[GYRO_FIX_IFLIGHTF722_TWING_10.1.1.5703.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/6986777/GYRO_FIX_IFLIGHTF722_TWING_10.1.1.5703.flx.zip)
try this and see if it fixes the issue

HrrrPippi, 2021-08-14:
Sorry mate
Already moved on. It's been 2 month, trying to get FLX to fly on my perfectly normal quad. I'm over wasting my time on flight one.

murph550, 2021-08-20:
> [GYRO_FIX_IFLIGHTF722_TWING_10.1.1.5703.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/6986777/GYRO_FIX_IFLIGHTF722_TWING_10.1.1.5703.flx.zip)
> try this and see if it fixes the issue

Didn't work. Same issue. Skipped motor config and only 3 spin.

PGIII, 2021-09-13:
> > [GYRO_FIX_IFLIGHTF722_TWING_10.1.1.5703.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/6986777/GYRO_FIX_IFLIGHTF722_TWING_10.1.1.5703.flx.zip)
> > try this and see if it fixes the issue
> 
> Didn't work. Same issue. Skipped motor config and only 3 spin.

So one motor isn't spinning ? are the other 3 working fine ? sounds like a different issue, do you have a link to the board you're using ?

DrGimbal, 2021-09-16:
Same problem here, 3 motors spinning, one not. With betaflight everythings fine...

murph550, 2021-09-17:
> So one motor isn't spinning ? are the other 3 working fine ? sounds like a different issue, do you have a link to the board you're using ?

Correct. https://shop.iflight-rc.com/quad-parts-cat20/electronics-cat27/succex-mini-fc-cat134/succex-d-mini-f7-2-6s-twing-fc-icm20689-pro993

PGIII, 2021-09-17:
> Correct. https://shop.iflight-rc.com/quad-parts-cat20/electronics-cat27/succex-mini-fc-cat134/succex-d-mini-f7-2-6s-twing-fc-icm20689-pro993

do you know which motor output the motor is on that's not spinning

DrGimbal, 2021-09-18:
My board is this here:
https://www.rctech.de/iflight-succex-d-f7-v2-1-twing-fc-hd-version.html

Motor Nr. 3 in the setup wizzard is not working