FlightOne/FalcoX-Issue-Tracker #126: H7 During Initial Setup VTX Power turned off right before Config askes to if you want to setup VTX power

2dogrc, 2021-01-31:
Firmware 4943 and on.

During initial setup with the H7, right after it asks if you would like to setup Auto level, then vtx power turns off.  The next step is to setup VTX power, but you can't because there is no power to the VTX.

WilidWilly, 2021-02-01:
If you go into the config and type RESETCONFIG does it do it the second time? That fixed it for me when I had this issue. Also to bypass this step for now, when the VTX shuts off just roll left, then it will finish the wizard and turn the VTX back on.

2dogrc, 2021-02-01:
Good question. 
  Roll left is what I did. This bug seems to come and go. So something I am starting to check more often. It would be terrible if this happened it the field at a different step and you could not finish setup.

closed 2021-06-10 by stale[bot]