FlightOne/FalcoX-Issue-Tracker #137: [NEW TARGET] New Target Request : CRAZYBEEF4

dustyprops, 2021-02-02:
Add new board target : CRAZYBEEF4

It would be awesome if I could fly my 1s whoop with FalcoX, been wanting to do this ever since.

PGIII, 2021-02-12:
Is it brushed or brushless ?

dustyprops, 2021-02-12:
> Is it brushed or brushless ?

Brushless.

binarylefmart, 2021-05-03:
I'm interested by this target too, CRAZYBEEF4FR, what is the status?

PieKiu, 2021-06-08:
Same here, if it helps in BF and Emuflight this target can also be flashed with MATEK411RX

PGIII, 2021-06-15:
> Same here, if it helps in BF and Emuflight this target can also be flashed with MATEK411RX. Tried flashing mine with FalcoX MATEKF411RX but no OSD comes on. If I flash MATEKF411 or MATEKF411SE OSD comes on and setup procedure starts. Problem is that the SPI RX is not recognised.

SPI RX will be more work than just adding a target since we don't have support for that at the moment, you're welcome to open a new feature request for it though

PGIII, 2021-06-15:
> I'm interested by this target too, CRAZYBEEF4FR, what is the status?

Whats the difference between that target and this one ? I plan on adding some more targets soon so this will probably be one of them

PieKiu, 2021-06-15:
> SPI RX will be more work than just adding a target since we don't have support for that at the moment, you're welcome to open a new feature request for it though

That's odd, because MATEKF411RX should share the same way to "speak" to the RX at least as far as I could tell. And the official page for the Matek F411-One states it has a FrSky SPI Rx on board.

Concerning your second question about the differences between CRAZYBEEF4 and CRAZYBEEF4FR
There are 4 differenct versions of this board:

- (Frsky version) CRAZYBEE F4FR PRO V2.0 Firmware target: CrazybeeF4FR

Frsky version Built-in SPI Frsky receiver with telemetry(D8/D16 switchable)

- (Flysky version) CRAZYBEE F4FS PRO V2.0 Firmware target: CrazybeeF4FS

Flysky version Built-in SPI Flysky receiver with telemetry(AFHDS/AFHDS2A switchable)

- (DSM2/DSMX version) CRAZYBEE F4DX PRO V2.0 Firmware target: CrazybeeF4DX

DSM2/X Version built-in Serial-based DSM2/DSMX compatible satellite receiver

- (No RX version) CRAZYBEE F4 PRO V2.0 Firmware target: CrazybeeF4DX

No RX version need to add external Receiver

I know this does not make things easier, anyhow I hope it covers what was asked for.

PGIII, 2021-06-15:
The MatekF411RX is just the target that is supported, the SPI Rx isn’t supported yet so people that use that target use a regular external receiver

PieKiu, 2021-06-15:
Good to know! Where would someone find such an important piece of information?