FlightOne/FalcoX-Issue-Tracker #142: Won't save config

ajp1078, 2021-02-06:
**Describe the bug**
A clear and concise description of what the bug is.
Runs through all wizards fine. Motors correct direction lay flat, on nose set up finishes says to reboot power cycle. 
Do so..... No change in Configurator screen and then wizards start again. I repeated this a few times still no saving of setup. 
Spoke to Nick Willard and tried the following.
RESETCONFIG
SAVE
msg #rest OK
Wizards launched again 
Did set up again
Before power cycle went into terminal and did following
SET wizard_flag=127
#ok
save
It bypassed the wizards I had the OSD screen up but would not arm or respond to switches on transmitter.
Tried that same process again SET wizard_flag=127
 got the following msg
#ok 
Then when I tried to 
save I got
#error save failed
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
MAC OSX 
Chrome latest version

**Additional context**
Add any other context about the problem here.

PGIII, 2021-02-08:
What FC is this and what does the error message show when you save ?

ajp1078, 2021-02-09:
Hey mate,

It is the lightning H7 when I hit save it says
#error save failed

First time I tried bypassing the wizards it did but wouldn't arm so
tried again a few times and kept getting the
#error save failed message

Thanks

PGIII, 2021-02-15:
Can you get a video of what youre doing when you see that message

ajp1078, 2021-02-16:
Hi Preston,

Yep I will get that through to you.

Cheers

proggod, 2021-04-11:
No updates so closing

closed 2021-04-11 by proggod

ajp1078, 2021-04-14:
I still haven't solved the issue. I have just had no time to get a good
video of the issue yet. So I will endeavour to get it to you asap

Regards