FlightOne/FalcoX-Issue-Tracker #143: Cannot save power setting of Smart Audio 2.1

Tonsuke201450, 2021-02-06:
**Version**
Version 10.1.0.5252

**Target**
Any Target

**Describe the bug**
Cannot save power settings of Smart Audio 2.1.  Any change will be reset to 25mw when exiting OSD setup menu.

**To Reproduce**
0. Configure hardware / software for Smart Audio 2.1 accordingly.
1. Go to OSD 6-13 menu.'
2. Change power setting.
3. Go to 13-13 and save / exit or just exit from OSD menu.
4. Will see VTX power setting reset to 25mw

**Expected behavior**
The power setting should be changed and saved effectively.

**Screenshots**
See youtube video below.
https://youtu.be/Hz0chWLgAWg

PGIII, 2021-02-08:
It gets that data from the VTX it doesnt actaully save anything, are you sure that VTX uses SA2.1 and not SA2.0?

Tonsuke201450, 2021-02-10:
Yes, it is working perfect with SA2.1 on Betaflight.  Also it doesn't work SA2.0 on FalcoX.  It looks like FC sending wrong control code to the VTX because you can check the exact moment that the value of vtx power changed 400 to 25 on OSD.  What is chaging the value if the FC is not sending anything at that moment. Addition to that, just connecting to the FC makes the vtx gone crazy and leds are started to brink randomly.  It never happen with Betaflight. 
https://www.youtube.com/watch?v=pNJoZxcFBuI

PGIII, 2021-02-15:
It only sends setting commands when you pick the send settings in the menu, outside of that it just polls for the VTX Info if it doesnt have it yet, im not sure what the different blinks on that VTX means so not exactly sure what the Video shows. But if you change the VTX settings and just pick exit does it still change power levels ?

Tonsuke201450, 2021-02-16:
Yes, it still goes back to 25mw even if I just exit without saving.  Again, it works perfect with BetaFlight and EmuFlight.  It is quite reasonable to assume something wrong with implementation of SA2.1 on FalcoX.

Tonsuke201450, 2021-02-16:
I bought AKK 25/200/400/600mW Switchable Mini VTX to replace EWRF VTXs but it doesn't work with FalcoX as well.  I can change if it is the first time after VTX detection.  It won't work after reset/power cycle.  Can you analyze protocol and fix it ?

Tonsuke201450, 2021-04-14:
It's not fixed yet.

closed 2021-06-10 by stale[bot]