FlightOne/FalcoX-Issue-Tracker #144: Throttle is not smooth, it's jumping in steps at quick stick movements

nasosfpv, 2021-02-06:
**Version**
10.1.0.5252

**Target**
Revolt OSD

**Describe the bug**
The throttle is not smooth as it should be, it's like it's moving in steps. This is noticable only at quick throttle stick movements though. It works perfectly fine when I downgrade to the latest beta. I don't know about previous alphas since I hadn't updated for a while.

ESC: Megabolt Extreme

WilidWilly, 2021-02-07:
What radio and rx?

nasosfpv, 2021-02-07:
> What radio and rx?

Tango 2 and nano rx

nasosfpv, 2021-02-07:
> > What radio and rx?
> 
> Tango 2 and nano rx

At the latest firmware

WilidWilly, 2021-02-09:
> > > What radio and rx?
> > 
> > 
> > Tango 2 and nano rx
> 
> At the latest firmware

What FW is the radio and rx on? And which UART is it hooked up to? and do you have any footage?

nasosfpv, 2021-02-09:
> > > > What radio and rx?
> > > 
> > > 
> > > Tango 2 and nano rx
> > 
> > 
> > At the latest firmware
> 
> What FW is the radio and rx on? And which UART is it hooked up to? and do you have any footage?

The radio is on 1.24 and crossfire on 4.10. It's hooked up to tx1 only, no rx3 soldered as per the diagrams I found.

 I'll get footage soon. The question is why does it work fine with the beta fw. I'd like to try more alphas to see from which point on the issue occurs, but I can't find any.

nasosfpv, 2021-02-10:
> The radio is on 1.24 and crossfire on 4.10. It's hooked up to tx1 only, no rx3 soldered as per the diagrams I found.
>
> I'll get footage soon. The question is why does it work fine with the beta fw. [...]

I got the footage https://youtu.be/BV11Z22tPAU

It's some sort of delay that you feel in rapid throttle movements.  In the third punch out I was quite scared because I had closed the throttle but it kept going up. Please share your thoughts on this with me, and if you think it's fixable. Thanks

nasosfpv, 2021-02-12:
Just realised that the stable version is newer than the beta, so I tried it and it has the same issue, maybe a little worse. It only works fine with the beta version.

Please tell me if there is any hope, because I'm about to give up and completely change setup. I'm also waiting for DJI so I can't stay with the beta version. 

In case it helps, my Megabolt Extreme is an 48hz only version, it doesn't go to 96hz, I guess it's was from an early batch.

PGIII, 2021-02-12:
If it worked on the earlier version and not on a later then im sure we can figure out the issue, will probably need to give you test bin so we can see whats going on exactly, do you only feel the delay in the throttle ? can you go in the diagnostic menu and go to the section where it shows the stick movements and get a DVR with moving the sticks around, also the latest alpha added a RX FPS OSD element, can you put that on and see what it shows its running at. Also DJI should work in the Current Beta version, MSP for DJI has been supported for a while now

nasosfpv, 2021-02-12:
> If it worked on the earlier version and not on a later then im sure we can figure out the issue, will probably need to give you test bin so we can see whats going on exactly, do you only feel the [...]

I can't easily tell about the other sticks because it was happening in rapid movements where throttle was always involved, I'll have to try again with that in mind.

I can't find anywhere the diagnostic menu, can you tell me where it is?  

I put the RX FPS on but I don't know what it's supposed to be normal so I uploaded the DVR for you here: https://youtu.be/pg98K6Mjk64

That test bin sounds nice, let me know how I can do that. Thank you!

PGIII, 2021-02-15:
It should be a constant number, if you have it set to 50hz it should be around 50fps, what speed do you have it set to? also i see you have a second OSD Is that coming from the VTX ? is the crossfire signal being passed through ?

PGIII, 2021-02-15:
[FLX_REVOLT_10.1.0.5332.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/5983580/FLX_REVOLT_10.1.0.5332.zip)
can you try this and see what the fps is

nasosfpv, 2021-02-15:
I had it set to dynamic so it should've been 150hz since I was in close range. I just tried it again with locked 50hz and the RX FPS behaves the same. 

The second osd is from the EVO Unify vtx. The nano rx is directly connected to the evo with ch.3 and 4 wires. 

So, I just tried the 5332 and the same issue happens. But now I noticed something that also happens with 5331 and I hadn't realised before. When I flash it fresh and before I run any config, the RX FPS is showing stable 50fps. But at this point, if I arm, the two back motors aren't spinning. The two front that are spinning sound great though, without any delay. As soon as I run the config wizard or just the radio config wizard, then the delay at rx fps comes back. But now all four motors are spinning. 

Also today, when I arm sometimes the RX FPS drop so low that I get a failsafe after a few seconds. This wasn't happening the previous days.

nasosfpv, 2021-02-15:
CORRECTION: It doesn't happen after I run the config, it happens after I power cycle for a second time. 

So to summarize, I flash it (with no battery connected), I unplug the usb, I plug in the battery and the rx fps are stable 50 but the two back motors don't spin. Then I unplug the battery, replug it and the delay is back but all four motors are spinning.

loutwice, 2021-02-17:
@nasosfpv  your problem looks like mine but on 50 hz i am having failsage ( mentionned here : https://github.com/FlightOne/FalcoX-Issue-Tracker/issues/91) 

what you said about throttle delay and steps it act the same way on 150hz

nasosfpv, 2021-02-17:
> @nasosfpv your problem looks like mine but on 50 hz i am having failsage ( mentionned here : #91)
> 
> what you said about throttle delay and steps it act the same way on 150hz

Sound like the same issue, the failsafes started occuring when I switched to 50hz where the delay was worse. 

The question is why does it happen only to us and everyone else is fine? Can you describe your setup in detail?

My setup: 
RevoltOSD
Megabolt Extreme 48hz version
nano rx on UART 1
Tango 2 pcb v3 
TBS EVO vtx directly linked to rx
6s 1900kv motors

loutwice, 2021-02-17:
> Sound like the same issue, the failsafes started occuring when I switched to 50hz where the delay was worse.
>
> The question is why does it happen only to us and everyone else is fine? Can you [...]

i think many dont have the problem because  many are still on older firmware - from what i know all the best pilot are still on the famous 2823 version ;-) which works fine no problem at all. 

My setup number 1 :                                my setup number 2 :                                       my setup number 3 :
RevoltOSD                                                RevoltOSD                                                        RevoltOSD
Megabolt                                                  AIKON                                                              Cant remember
nano rx on UART 1                                    nano rx on UART 1                                         nano rx on UART 1
Tango 2 pcb v1                                         Tango 2 pcb v1                                               Tango 2 pcb v1
 TBS VTX directly linked to rx                   VISTA                                                               TBS directly linked to rx
4s 2750 kv                                                  6s 1950kv                                                       4s FPVCYCLE 3600kv 

i am making a new setup with the H7 will see if the problem is there or not 

@WilidWilly and @PGIII  as i told you i can test all previous version and so we can find since which version the behavior happen , will just need firmware file.   like we know 10.0.6.2823 is the best working one, give me the firmware from half this one and half newest ( dichotomic search ;-) )

PGIII, 2021-02-18:
Theres definitely  a lot of people on versions past 2823;P doesnt seem to be a wide spread issue with CRSF, likely has something to do with having the RX connected to the VTX directly since you both have them connected, but youre issues arent exactly the same, we will have to give you guys some more test FW to narrow down whats causing your issues

nasosfpv, 2021-02-18:
> Theres definitely a lot of people on versions past 2823;P doesnt seem to be a wide spread issue with CRSF, likely has something to do with having the RX connected to the VTX directly since you both [...]

On the second setup of @loutwice, the rx isn't connected to a vtx because it's a vista, but I'll give it a try and disconnect mine. To me revolt osd with crsf seems to be the common factors.

Also, I realised now that just connecting it to the configurator, disconnect and then plug in a lipo for the first time makes it go in that state where the RX FPS are stable at 50 and the two back motors aren't spinning. Then, when I power cycle again, the usual delay starts again and all motors are spinning. @loutwice maybe you can try it and see if the same happens to you too.

@PGIII I can do as much testing as it takes, it'd be nice to figure it out.

loutwice, 2021-03-12:
tested last alpha - problem still persist

krunked-fpv, 2021-03-12:
This makes me think this is a RX issue with your quad

loutwice, 2021-03-13:
> This makes me think this is a RX issue with your quad
> […](#)

well it will not work then with old Falcox version if it was a problem from my  RX .. and problem on 3 of my kwads and revoltOSD .. version 2823 works flawlessly . so there might have something wrong somewhere ( from my side or firmware side ) .

I woud like to try all version from 2823to last and find from which version the problem occurs , is there any way to do so?

nasosfpv, 2021-03-13:
wow I just flashed 10.1.0.5414 and the problem is gone! RX FPS and TX FPS are stable at 150

@loutwice which one did you flash? There was 10.1.0.5407 in the meantime, I didn't try that one

@PGIII

loutwice, 2021-03-16:
@nasosfpv  i tried 5414 and on 50 fps still have failsafe , i really don't understand what is happening

nasosfpv, 2021-03-16:
> @nasosfpv i tried 5414 and on 50 fps still have failsafe , i really don't understand what is happening

Hey man I got good news I think. It wasn't the latest alpha that made the difference as I thought. It's something about the ESC telemetry that messes it up. You have to disable it through the osd if possible or remove the connection from esc completely and test again.

A few weeks ago I changed ESC because of some other issues and I never tested it with any alpha, I just left 2823 on it. 
I put on a Hobbywing 60A which has no esc telemetry, just a current sensor which is not working as it is connected, and this is what (I think) fixed the problem after all.  Disabling the esc telemetry was also the fix for another friend of mine. Check it and report back!

@loutwice @PGIII @WilidWilly

PGIII, 2021-03-16:
> > @nasosfpv i tried 5414 and on 50 fps still have failsafe , i really don't understand what is happening
> 
> Hey man I got good news I think. It wasn't the latest alpha that made the difference as I thought. It's something about the ESC telemetry that messes it up. You have to disable it through the osd if possible or remove the connection from esc completely and test again.
> 
> A few weeks ago I changed ESC because of some other issues and I never tested it with any alpha, I just left 2823 on it.
> I put on a Hobbywing 60A which has no esc telemetry, just a current sensor which is not working as it is connected, and this is what (I think) fixed the problem after all. Disabling the esc telemetry was also the fix for another friend of mine as I found out today. Check it and report back!
> 
> @loutwice @PGIII @WilidWilly
So you're saying if you disable the ESC telemetry it works fine ? was the ESC telem working ?

nasosfpv, 2021-03-16:
@PGIII Yes this is what I'm saying. Previously, Esc telemetry was connected from the megabolt to the revoltosd, but I don't remember if it was actually working because I don't care much for that stuff. But now the Hobbywing 60A doesn't even have esc telemetry and the problem is gone. And this also confirms my other friend who also solved this by disabling esc telem.

nasosfpv, 2021-03-16:
@PGIII To make myself clear in case I confused you, since my english aren't the best:

What fixed it for me was replacing the Megabolt Extreme with a Hobbywing 60A Esc. Now it works fine with all alphas. But the key difference between these two escs is that the hobbywing doesn't have esc telemetry. In addition, I spoke with a friend that had the same issue and he solved it by disabling the Esc telemetry. So I definitely think this is worth a try

PGIII, 2021-03-16:
Ill have to see if I can replicate the issue, but glad your got it working

loutwice, 2021-03-16:
> Ill have to see if I can replicate the issue, but glad your got it working

@PGIII @nasosfpv  how that is a good news - let see if it works I can easily do it let me one hour !

loutwice, 2021-03-16:
@WilidWilly @PGIII @nasosfpv 

maaaannnn i confirm ! Deactivating blheli telemetry just fixed my problem too !

proggod, 2021-04-11:
Should be fixed

closed 2021-04-11 by proggod