FlightOne/FalcoX-Issue-Tracker #145: Vifly v2 Buzzer makes weird sound on Alpha 15 (10.1.0.5252) for the Lightning H7

marcelinojorgeromero, 2021-02-07:
**Version**
SKITZO_LIGHTNING 10.1.0.5252

**Target**
SKITZO_LIGHTNING H7

**Describe the bug**
I plug the battery and the buzzer starts to make weird noise, at the end is like is dying slowly.

**To Reproduce**
1. I updated the firmware to the latest from Alpha 14
2. Plugged the battery and started making weird sounds
3. Also the switch wasn't working (was working before)

**Expected behavior**
Buzzer was making a standard sound always.

**Screenshots**

https://user-images.githubusercontent.com/2983268/107146064-81104b80-6924-11eb-8b93-14632b26d27f.mp4

PGIII, 2021-02-07:
That’s the startup tone for the buzzer; but what happens when you try to use the buzzer mode to turn it on?

marcelinojorgeromero, 2021-02-07:
> That’s the startup tone for the buzzer; but what happens when you try to use the buzzer mode to turn it on?

That's the first time i hear that sound. The buzzer wasn't working at all.

PGIII, 2021-02-08:
seems to be working fine on a regular buzzer, can you get me a dump from the FC

marcelinojorgeromero, 2021-02-10:
> seems to be working fine on a regular buzzer, can you get me a dump from the FC

how can i do that?

closed 2021-06-10 by stale[bot]