FlightOne/FalcoX-Issue-Tracker #161: no armaba iflight f405

joanlopezcorredor, 2021-02-28:
INGLES:
everything well I detect everything I install, but I did not arm myself, not knowing how to fix it, try to put the firmware betaflight f4, and when trying that with the same fc and number of fc and liscience, the fc has died does not detect fc tampco mode dfu, pass impulse rc and nothing, no and I could spend it, at night I received code lyscience for this fc but died before trying it , I need help or to be given the liscience for another fc to be able to test, because if it works I want liscience for all my fc.
ESPAÑOL:
todo bien detecto todo instalo, pero no me armaba , al no saber como poder solucionarlo, intente poner el firmware betaflight f4, y al intentar eso con la misma fc y numero de fc y liciencia, la fc ha muerto no detecta fc tampco modo dfu, pase impulse rc y nada, no e podio gastarla, a noche recibi codigo liciencia para esta fc pero murio antes de probarla, necesito ayuda o que me dieran la liciencia para otra fc para poder probar, ya que si funciona quiero liciencia para todas mis fc.

PGIII, 2021-03-08:
Im not sure i understand are you saying youre having issues arming or flashing ?

joanlopezcorredor, 2021-04-14:
una pregunta la fc me murio, porque estaba ya tocada. la liciencia la pierdo imagino. no hay algun tiempo minimo por casos de que pueda dar fallo una fc en muy poco tiempo, y podais dar liciencia a otra . o se pierda aun siendo incluso el mismo dia por un corto o alguna circustacia? . pierdo la liciencia? es mi pregunta .

closed 2021-06-10 by stale[bot]